pcdshub/lcls-twincat-vacuum #14: .tmc file does not fully render enums for E_PressureState

n-wbrown, 2020-02-28:
When a project depending on L2SI-Vacuum-Lib is built, the process fails to render the `<Enum>10</Enum>` field for `<EnumInfo>` tag in the datatype specifications area. 

The XML tag can be added manually producing successful pytmc-build .db files.

We suspect this is caused by this specific value of this specific enum being defined by an externally specified value:
https://github.com/pcdshub/L2SI-Vacuum-Lib/blob/7653d1ad42a26be8875d95d74a5f048be6eb2b5d/L2SIVacuum/DUTs/Gauges/E_PressureState.TcDUT#L7

The value is specified here:
https://github.com/pcdshub/L2SI-Vacuum-Lib/blob/7653d1ad42a26be8875d95d74a5f048be6eb2b5d/L2SIVacuum/GVLs/Constants.TcGVL#L6

@klauer 
@slacAWallace

n-wbrown, 2020-06-13:
@ghalym Thanks! The IOC build process is much smoother now!

ghalym, 2020-06-13:
Great to hear that! Sorry it took a while.

closed 2020-06-16 by ghalym