pcdshub/lcls-twincat-vacuum #18: Setpoint limits should be global variables

slacAWallace, 2020-03-14:
Various FB setpoint limits should be parameterized, made global, and pytmc'd so they can be adjusted on a system by system basis.

ghalym, 2020-03-14:
Setpoints are parametrized and pytmc'd in the devices DUTs and are available through expert screens. They are not global variables so as not to be hard coded in the program. What is hard coded is the Maximum allowed setpoint. e.g the PIP checks if the user setpoint is higher that 1e-4 and rejects it, same applies for some turbos, no backing pressure set point should go higher than a certain value (MaxBackingPressure).

slacAWallace, 2020-03-14:
Right, I think the maximums should be made into globals that we can adjust from EPICS.

ghalym, 2020-03-14:
The maximum values are per the device manual. Its the recommended value, should the user really be allowed to change those values. What kind of control or check are we doing to make sure they don't set a value that could damage the device.

slacAWallace, 2020-03-14:
So this is coming from increasing the ion pump interlock setpoint adjustment. Right now we'll have to cut a new release of the lib to increase that upper limit. And I anticipate different systems might prefer different maximums. But maybe this is overkill...

ghalym, 2020-03-14:
For the specific case of the ion pumps, the manual specifies a pressure that is at 1e-4 or lower. We can change the FB so that the ion pumps turn on with the GPI or GCC but switches off based on its own pressure reading. 
And I agree, different systems will have different maximums, and those maximums should be set in the PLC once for every device based on the system's engineer recommendations. 
And sure we can always remove those setpoint checks. My worry is that the interlock rely on correct setpoints to protect the devices.

slacAWallace, 2020-03-14:
Ahh I did not know the manual said that... that's critical. We'll need to rethink gauges.... the pirani or the interlock for the PIPs isn't working (they keep tripping off).