pcdshub/lcls-twincat-vacuum #21: Gauges/Pumps that are interlocked should pass to EPICS the name of the interlocked component so they can be seen on the expert screens for debugging

slacAdpai, 2020-03-18:
Gauges/Pumps that are interlocked should pass to EPICS the name of the interlocked component so they can be seen on the expert screens for debugging.

ghalym, 2020-03-18:
I am not sure I understand. If a component is interlocked, the interlock state shows on the widget. Do you wanna show the reason why this device is interlocked?

ghalym, 2020-03-18:
Its not always that straight forward. Most of the devices have an external interlock bit, that can be interlocked with any logical expression.  That applies to VGCs as well.
My initial thought was that there will be confluence page that shows the interlock table and interlock conditions for every device that is defined by the system's engineer.
For straight forward interlocks, I will see how to include that.

slacAdpai, 2020-03-20:
After talking with Alex, he said that is would be a great idea to add clarity to the vacuum system for the user. Can we add another element in the structure to add the string name of the structure?

ghalym, 2020-03-20:
What do you mean name of the structure?

closed 2020-06-16 by ghalym