pcdshub/lcls-twincat-vacuum #25: Duplicate definition of variable 'fbLogger' in function block 'FB_PIP_Gamma' and in base 'FB_Pump' [lcls-twincat-vacuum, 0.0.0 (slac)]	FB_PIP_Gamma

klauer, 2020-03-20:
https://github.com/pcdshub/lcls-twincat-vacuum/blob/cae969c79ee5732afe7d2d9c32427fca54ff7acc/L2SIVacuum/POUs/Functions/Pumps/FB_PIP_Gamma.TcPOU#L30

https://github.com/pcdshub/lcls-twincat-vacuum/blob/cae969c79ee5732afe7d2d9c32427fca54ff7acc/L2SIVacuum/POUs/Functions/Pumps/FB_Pump.TcPOU#L12

klauer, 2020-03-20:
Also: `Duplicate definition of variable 'tAction' in function block 'FB_PIP_Gamma' and in base 'FB_Pump' [lcls-twincat-vacuum, 0.0.0 (slac)]	FB_PIP_Gamma`

klauer, 2020-03-20:
Hmm, `tAction` is `F_TRIG` in `FB_PIP_Gamma` but `R_TRIG` in `FB_Pump`

ghalym, 2020-03-20:
Good Catch! Thanks!
I created a base class with all the logger variables, so that I won'to go on copy and paste in every pump function block. Forgot that it was in the pip already.
Still need more work to properly make it oo and modular.

closed 2020-06-16 by ghalym