pcdshub/lcls-twincat-vacuum #29: FB_PTM_Agilent should be updated to resemble FB_PTM_TwistTorr

slacAdpai, 2020-03-27:
I noticed from reading the TV301 and TV701 manuals that they look about the same as the TwisTorr manual. Can FB_PTM_Agilent be updated to look more similar? Maybe this was already done, I saw it checked on the ECP Trello checklist.

ghalym, 2020-03-27:
The FB_PTM_Agilent, is for rack mount controllers. It works with the ones on the GEMs in the FEE.  And you are correct the twsitorr function block is better for the TV301 and TV701 onboard controllers. So, I will deprecate the twsitorr function block and make a new function block for onboard controllers.
Any suggestions what to call it, FB_Agilent_???

slacAdpai, 2020-03-27:
If there is no difference between the twistorr function block and TV301 and TV701, should we just have one function block for all these agilent turbo pumps? Then we would only need to maintain support for one agilent turbo pump function block. I think FB_PTM_Agilent works.

ghalym, 2020-03-27:
FB_PTM_Agilent, is for the rack mount versions of these controllers and they are different from the on board ones.
The on board one is the same as the twistorr.

slacAdpai, 2020-03-27:
If the rack mountable controllers and onboard controllers are different, then FB_PTM_Agilent_Rack and FB_PTM_Agilent_Onboard?

slacAdpai, 2020-03-27:
When there is time, can the Vacuum architecture get updated as well? I have been using it a lot and it is very helpful in designing the Namaste vaccuum system!