pcdshub/lcls-twincat-vacuum #35: PLC message spam freezes TwinCAT

n-wbrown, 2020-06-11:
Messages from the MKS275 (Integrated Pirani) build up in the TwinCat Messages inbox at a very high rate (4.1k/minute in the SXR Vacuum system). TwinCAT performance progressively degrades with more messages and becomes completely unresponsive around 30k messages. 

The clear messages button needs to be pressed every 5 minutes that the TwinCAT project is open to prevent TwinCAT from becoming uselessly slow. 

This looks suspiciously similar to the logger spamming issue we encountered recently, only redirected to the TwinCAT messages system rather than logstash. 

[image]

@ghalym 
@slacAWallace

ghalym, 2020-06-11:
I checked those gauges on the screen, the adjacent GCC reads -10, which means that this pirani should be fixed at 1-4. However it keeps changing. My suggestion is that the gauges needs calibration according to the screen capture below.
[image]
I also noticed that the MKS317 RTDSK0, TV2K0-1 TV2K0-2 and PA1K0 are not calibrated, while the GCC reads good vacuum those piranis are reading at -3 level, which indicate that they have not been calibrated from the controller as well.
I can implement hysteresis on MKS275, but that would just mask the issue. I would suggest that we calibrate those gauges first MKS317 and MKS275, and then implement the hysteresis.
As for the the spam of messages, if we have a truly faulty gauge or signal it might spam the logger, I suggest we check with the logger system if there is a way to put a maximum update rate per device or system. 

@n-wbrown 
@slacAWallace 
@slacAdpai

slacAWallace, 2020-06-11:
I am implementing a circuit breaker for the logger right now. Should be released this morning and be compatible with an online change (if you've been running the latest release).

Gauges need to be calibrated properly, but I think a condition that detects the symptom and emits a single warning (maybe periodically), is the correct way to indicate that.

slacAWallace, 2020-06-11:
@klauer What does this do:
https://github.com/pcdshub/lcls-twincat-general/blob/4f45e54ff29676050368109dffca0f340c7054ac/LCLSGeneral/LCLSGeneral/POUs/Logger/FB_Listener.TcPOU#L19

klauer, 2020-06-11:
@slacAWallace it outputs the JSON message to visual studio.

n-wbrown, 2020-06-12:
@ghalym @slacAWallace.

We went through the calibration today. I think the noise is normal and hysteresis of some degree needs to be implemented.

The manual advises calibrating the low-pressure-boundary to .375 volts. The function block for the gauge is programed such that the voltage must be lower than .375 volts to count as ValidLo. Naturally there is some noise in the signal on the order of E-2 volts (possibly more). On a gauge in the ValidLo range, the signal calibrated to average at .375 will rapidly fluctuate between the Valid and ValidLo regime.

As a band-aid for this message spamming issue, I have had the gauges tuned slightly low to .370 V. This does not fully mitigate the issue and I think should not be counted on as a permanent solution. 

I think the quantity of noise observed is normal and healthy and we should modify the function block to accommodate for this noise. Attempting to resolve the issue with significant re-calibration of the gauge will require us to revisit the voltage-to-pressure calculation. 

I am now somewhat concerned that gauges not experiencing this rapid state toggling may be poorly calibrated and providing inaccurate pressure readings as a result.

ghalym, 2020-06-12:
My suggestion was not to merely mitigate the spamming issue, but to ensure that we are getting proper reading out of the gauge by calibrating it as per the manual and thus a proper pressure reading. The gauge should also be calibrated at atmosphere. Calibration of the gauge, as per manual, ensure that we get accurate readings.
[image]

Now, I will check the archiver that has the AI raw signal from the gauges. And will see what hysteresis band and thresholds are appropriate based on the calibrated signal noise.

ghalym, 2020-07-02:
@n-wbrown Now that the gauges have been calibrated and I have included a deadband on the ValidLo to Valid state switch, does this still seem to be an issue?