pcdshub/lcls-twincat-vacuum #42: PIP Function Block- Should AI_Offset and Auto_on be PVs with read and write access or only read acess?

slacAdpai, 2020-06-19:
IN the PIP Function block, currently AI_Offset and Auto_On PVs  are read access only. Should the User be able to change the values accordingly? Especially being able to toggle auto on?

slacAdpai, 2020-06-19:
I believe that the analog offset set to 13 is working well. I will finalize the results tomorrow when I ask someone to change the offsets on the QPC to 13.

n-wbrown, 2020-06-22:
@slacAdpai @ghalym. Are we standardizing 13 as the new offset or are we keeping 11? If so I'll update QPC controller setup guide to match (https://docs.google.com/document/d/1Hc8TfKOTtBE7kt8lk-NOo_f31xrU7EGBmc2bz3cXqt8/edit#).

slacAdpai, 2020-06-22:
@n-wbrown @ghalym @jyin999 @jyotiphy Using the new library, 13 has been working very well. It gives us more precise pressures in the 1e-11 and lower range. It is also easier to tell if the ion pump controller is acting up. If everyone is ok, I would recommend standardizing to 13.

jyin999, 2020-06-24:
I didn’t compare to other offset.  I use 13.  It gives about 15% difference between PLC and QPC controller at e-10 Torr level.

ghalym, 2020-07-02:
I have changed the pramgas to be of type io instead of just i.