pcdshub/lcls-twincat-vacuum #47: FB_VFS vacuum fault ok does not seem to update when Reset vacuum fault button is pressed

slacAdpai, 2020-06-24:
FB_VFS vacuum fault ok does not seem to update to fault ok when Reset vacuum fault button is pressed. Instead the vacuum fault ok bit only updated to fault ok when the fsv is open. Is this correct?

ghalym, 2020-06-25:
```
(* Neighboring Valves Interlock bit is reset When the Vacuum event error is reset and the Shutter is opened *)
IF (NOT i_xTrigger) AND (NOT iq_stValve.xERR_ExtFault) AND i_xOpnLS THEN
	q_xVAC_FAULT_OK := TRUE;
END_IF
```

Yes, this is how it is implemented. This requirement was included because there ins't always a gauge between the FSV and the neighboring VGC. And we didn't want to open the VGC without knowing the pressure on the side of the trapped volume. And since FSV do not function as complete isolation valves, it was agreed that it is safer to open the FSV to that trapped volume.  

So q_xVAC_FAULT_OK  that should be linked to the VGCs external interlock bit, will remain false until the FSV is open.

ghalym, 2020-07-02:
Hi Allen, does this seem to be still an issue?

slacAdpai, 2020-07-02:
Not any more. All good.

closed 2020-07-02 by slacAdpai