pcdshub/lcls-twincat-vacuum #50: Tools for staggered startup interlock or procedure

n-wbrown, 2020-06-30:
@ghalym 
@slacAWallace 
@slacAdpai 
@sfsyunusSLAC 

Given the spatial restrictions of LAMP, there is a significant chance that we may need this as a protection for the DC system. Through TMO's development, the idea of staggering the startup of turbo pumps has been repeatedly raised due to the restrictions on available power. While it is possible that testing proves that this is not ultimately required, I think we should start the conversation now in case it is. 

I think it is worthwhile to standardize this type of procedure(s) in the vacuum library. Given that this is used to protect other systems, I think it makes sense to place in the PLC. This is a system specific restriction and isn't something we should rely on an operator or script-writer to remember each time they work with such a system.

n-wbrown, 2020-06-30:
To kick this off, I propose we implement an interlock-like function that limits whether a pump may turn on depending on the number of other pumps in a starting state.

This would create relatively few restrictions while still limiting the number of pumps that may be simultaneously starting.

It would probably take this form: 
```
FUNCTION F_TurboStaggeredStartup : BOOL;
VAR_INPUT
   TargetPump:  ST_PTM; // Only restrict this pump if it is in the starting state. Do not restrict it if it is already at maximum speed.
   MaxStarting: INT; // The maximum number of pumps that may be in the starting state at any given time. 
END_VAR
VAR_IN_OUT
    PumpPool: ARRAY [*] OF ST_PTM; // List of pumps in a group. From this group only [MaxStarting] shall be permitted in the starting state simultaneously.
END_VAR
```

ghalym, 2020-07-02:
That looks like the way to do it. 
So the plan is to link the output of this function to the external interlock bit the of the PTM function block?

n-wbrown, 2020-07-02:
Using this function's output as the external interlock bit is what I was thinking of. We would probably AND it with the standard interlock function to ensure that both conditions are met.